beagleboard/BeagleBoard-DeviceTrees #24: SPI CS configuration for P9.23 on AM572xx, kernel 4.19

dmconover, 2020-10-23:
I have been working on adapting an industrial application from the BeagleBone Black to the AI and came across the ``v4.19.x-ti-overlays`` branch of the Beagleboard Device Trees repo. Thanks to the great work done here already, I was able to properly configure CAN,UART,etc fairly easily! However, one necessery step I need is to configure P9.23 as a SPI chip select pin. According to the [BeagleBone Cape Header Spreadsheet](https://docs.google.com/spreadsheets/d/1fE-AsDZvJ-bBwzNBj1_sPDrutvEvsmARqFwvbw_HkrE/edit#gid=1518010293) (cell BT58, specifically), this should be an option for pin configuration. However, this doesn't seem to be an option in ``am572x-bone-common-univ.dtsi``, and if I try to add it myself, the resulting overlay will brick my device. I was wondering if someone better acquainted with this project had any ideas about how to add this functionality into the ``v4.19.x-ti-overlays`` branch. The resulting option would appear as something like:

  - 	P9_23_spi_cs_pin: pinmux_P9_23_spi_cs_pin { pinctrl-single,pins = <
		P9_23( PIN_OUTPUT_PULLUP | INPUT_EN | MUX_MODE3)>; };

If someone could even point me in the right direction, I'd be happy to try to implement it myself!

Thanks,

Doug

lorforlinux, 2020-10-25:
@dmconover Hey, I have submitted PR #26 please take a look and let me know how it goes :)

closed 2020-10-25 by jadonk

dmconover, 2020-10-26:
@lorforlinux Thanks so much for picking this up! Unfortunately I am still having issues. I pulled your changes and built BBAI-SPI0_1.dtbo, and included it in my uEnv.txt. My BBAI won't boot after these changes.. I'm wondering if I have an out of date image/kernel, can I ask which one you're using? On my end:

``debian@beaglebone:/var/lib/cloud9$ uname -r``
``4.19.94-ti-r50``

uEnv.txt:
###########################################################################3
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.94-ti-r50
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/boot/dtbs/BBAI-SPI0_1.dtbo
#uboot_overlay_addr1=/boot/dtbs/BONE-CAN1.dtbo
#uboot_overlay_addr2=/boot/dtbs/BONE-UART4.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=<file4>.dtbo
#uboot_overlay_addr5=<file5>.dtbo
#uboot_overlay_addr6=<file6>.dtbo
#uboot_overlay_addr7=<file7>.dtbo
###
###Custom Cape
#dtb_overlay=<file8>.dtbo
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

##enable x15: eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3-no-eeprom.sh

######################################################################

The base image I am using is linked in [this discussion](https://groups.google.com/g/beagleboard/c/-7xASl-oqic)

lorforlinux, 2020-10-26:
@dmconover I would recommend you to follow the instructions I have provided on the official site for installing the compatibility layer, https://deepaklorkhatri.me/GSoC2020_BeagleBoard.org

First you have to follow the steps for the eMMC image then for the SD Card image. Updating the eMMC image is crucial!

Changes for your uEnv.txt, you don't need to include the path just put the name. The changes has been merged so this overlay (BBAI-SPI0_1.dtbo) should just work out of the box now (no need to compile). To make sure you have the overlay you can check in `/boot/dtbs/4.19*/overlays/`.

Example,

```
...
uboot_overlay_addr0=BBAI-SPI0_1.dtbo
...
```

RobertCNelson, 2020-10-26:
@dmconover any chance do you have a 3v3 ftdi adapter?  it would be useful to see what u-boot is doing..

https://www.digikey.com/en/products/detail/digi-key-electronics/BBCAI/10187731

https://www.digikey.com/en/products/detail/ftdi-future-technology-devices-international-ltd/TTL-232R-3V3/1836393

Regards,

dmconover, 2020-10-26:
@lorforlinux, I will make a fresh image using your procedure and get back to you. @RobertCNelson I will go check to see if I have one in the electronics lab, if not, I will order one.

lorforlinux, 2020-10-26:
@dmconover I think you can skip the installation steps if other overlays are working for you. The problem is that the base DTB (am5729-beagleboneai.dtb) is not updated and to update that you have to clone this repo and run `make install`, that will resolve your problem.

dmconover, 2020-10-26:
Ah! That was it! Thank you, that was the issue, it's loading the overlay properly now and the pins are configured correctly. Thank you so much for your patience!

lorforlinux, 2020-10-26:
It's alright, Just spread the word about the new Compatibility layer and if possible contribute with documentation or code. There are different sections for each Bone Bus on https://deepaklorkhatri.me/GSoC2020_BeagleBoard.org if you could contribute to them OR share the link to anybody interested in contributing that would be awesome :)

dmconover, 2020-10-26:
Can do, I'll take a look and see how I can help out

PR #26 (2020-10-25): Add BBAI-SPI0_1.dts (Fix #24)
@RobertCNelson @jadonk Please review my code :) Everything has been tested on BBAI.

This PR includes DT overlay for enabling channel@1 of bone_spi0 (only available on BBAI) which fixes #24
commit 39b14b81 (2020-10-25): Merge pull request #26 from lorforlinux/fix#24

Add BBAI-SPI0_1.dts (Fix #24)