beagleboard/BeagleBoard-DeviceTrees #27: Convert to panel-simple

RobertCNelson, 2020-10-29:
https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/456d8d36d4226f7375aef1c4dffc19fdfc09dc11/src/arm/bbb-bone-buses.dtsi#L652-L670

lorforlinux, 2020-11-24:
Thanks, @RobertCNelson Will try it out :)

Siddhant-K-code, 2021-01-31:
Maintainers, I guess, We Can Close this issue now!

lorforlinux, 2021-02-01:
Not able to use it successfully yet, 
I can't recommend closing it until we have a consistent working demo.

RobertCNelson, 2021-02-01:
@Siddhant-K-code you didn't include your *.patch...

harshavardhan-b20, 2023-10-01:
Hello,

I'm excited about contributing to this issue and believe it aligns well with my skills and interests. Could you please assign me to work on this? I'm ready to dive in, discuss the details, and start contributing.

Thank you!
Harshavardhan Bellamkonda