beagleboard/BeagleBoard-DeviceTrees #29: RE: BBAI, Compatibility Layer, and Setting Up/PWM

silver2row, 2020-11-20:
Hello,

I followed the instructions here: https://deepaklorkhatri.me/GSoC2020_BeagleBoard.org/

- First, I followed those instructions with this image and kernel:
    BeagleBoard.org Debian Buster IoT Image 2020-11-16
    4.19.94-ti-r56

- Secondly, I looked for pwm in /sys/class/pwm/, /dev/pwm/, and /dev/bone/pwm/. 
    I have not been able to get the pwm peripheral located yet

...

Please advise me on what to tackle next. I have tried to clone w/ git many things.

https://github.com/beagleboard/bb.org-overlays

and...

https://github.com/lorforlinux/bb.org-overlays

and...

https://github.com/beagleboard/BeagleBoard-DeviceTrees

I tried make, adding BONE-PWM1.dtbo to /boot/uEnv.txt, and I even tried different git checkout commands.

Is there an easier way to test these Bone_Bus capabilities? I am asking b/c I have a BBAI and I have been unable to make this work just yet.

Seth

RobertCNelson, 2020-11-20:
Please include output off..

sudo /opt/scripts/tools/version.sh

silver2row, 2020-11-20:
Hello Sir,

I know I seem to wonder a lot. I got the /sys/class/pwm/pwm-2:0 interface to work w/ the BBAI so far on p9.14.

Anyway...here is the output of sudo /opt/scripts/tools/version.sh:

https://pastebin.com/6Z06D9Cf

Seth

P.S. Please let me know if you need anything else.

silver2row, 2020-11-20:
Hello Sir,

Seth here. Um, I used a lib. in python to handle PWM onboard the BBAI so far. It worked!

...

Anyway, I do not see the entry in /dev/pwm/ yet. I will keep trying. 

Seth

P.S. Please just let me know if you need any more info. from me.

RobertCNelson, 2020-11-20:
/sys/class/pwm/ is the correct location

silver2row, 2020-11-21:
Hello,

Thank you for replying. I got it up and running w/ a pwm instance doing the on/off routine for p9.14 on the BBAI. I am learning slowly about this new idea.

Seth

P.S. I can close this issue if it is of no concern...let me know.

closed 2020-11-21 by silver2row