beagleboard/BeagleBoard-DeviceTrees #30: Overlay for GEN4-4DCAPE-43CT-CLB not working for BBAI

mjr1ch, 2020-11-24:
Hi 

I have the compatibility  layer and appropriate overlay installed. Show pins has the proper pins slated for the cape. Also modification to the cape adapter board was made to allow power from the USB port since the BBAI lacks a barrel jack. When powered the Power LED stays on but I have no activity otherwise. This behavior only occurs when the cape is attached. If the cape is removed the BBAI functions as normal. 

Please advise and thanks in advance.

lorforlinux, 2020-11-24:
Hey, 

you can do it in 3 simple steps after configuring the jumper on the board so that it uses power from the board.

#### Step 1

Make sure you have up to date compatibility layer. follow the instructions on the homepage of the [Compatibility layer's official site](https://deepaklorkhatri.me/GSoC2020_BeagleBoard.org/) for that.

#### Step 2

Update overlays,

1. git clone https://github.com/beagleboard/BeagleBoard-DeviceTrees.git
2. cd BeagleBoard-DeviceTrees
3. sudo make install

*you probably don't need it after updating the board (compatibility layer).

#### Step 3

Update the uEnv.txt on BBAI, on BBB it's automatically loaded.

```
...
enable_uboot_overlays=1
uboot_overlay_addr0=BB-BONE-4D4C-01-00A1.dtbo
...
```

Siddhant-K-code, 2021-01-31:
Maintainers, I guess, We Can Close this issue now!

mjr1ch, 2021-01-31:
I worked with Deepak and I would not necessarily consider it closed. The cross compatible overlay did not work for me. I did get the cape that gets auto detected by BBAI to work.